Project-Babble/Baballonia #87: [Bug]: Entire program crashes ~1second after startup

virtuallyaverage, 2025-09-07:
### Version Number

latest (v1.0.0.1 as of writing)

### Describe the bug

Installation reports successful.

Start program, closes immediately after opening. 

This happens on: 

 - release version
 - Release locally built from main 
 - `Baballonia.Desktop` in Rider's Debugging mode.

### Steps to reproduce

# `Baballonia.Desktop` in Rider's Debugging mode mentioned earlier;

 - `git clone git clone https://github.com/Project-Babble/Baballonia.git --recursive`
 - `rider Baballonia/Baballonia.sln`
 - Select `Baballonia.Desktop` target 
 - Run `Baballonia.Desktop` (ctrl+F5)

Crashes with trace:
```
Unhandled exception. System.AggregateException: One or more errors occurred. (Value cannot be null. (Parameter 'logger'))
 ---> System.ArgumentNullException: Value cannot be null. (Parameter 'logger')
   at System.ThrowHelper.Throw(String paramName)
   at System.ThrowHelper.ThrowIfNull(Object argument, String paramName)
   at Microsoft.Extensions.Logging.LoggerExtensions.Log(ILogger logger, LogLevel logLevel, EventId eventId, Exception exception, String message, Object[] args)
   at Microsoft.Extensions.Logging.LoggerExtensions.Log(ILogger logger, LogLevel logLevel, String message, Object[] args)
   at Microsoft.Extensions.Logging.LoggerExtensions.LogError(ILogger logger, String message, Object[] args)
   at Baballonia.Services.LocalSettingsService.ReadSettingAsync[T](String key, T defaultValue, Boolean forceLocal) in D:\Github-projects\Babble\Baballonia\src\Baballonia\Services\LocalSettingsService.cs:line 86
   at Baballonia.Views.OnboardingView.<>c.<<ShowIfNeeded>b__7_0>d.MoveNext() in D:\Github-projects\Babble\Baballonia\src\Baballonia\Views\OnboardingView.axaml.cs:line 48
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at Baballonia.Views.OnboardingView.ShowIfNeeded(Window parent) in D:\Github-projects\Babble\Baballonia\src\Baballonia\Views\OnboardingView.axaml.cs:line 45
   at Baballonia.Helpers.MainWindowExtensions.ShowOnboardingIfNeeded(Window mainWindow) in D:\Github-projects\Babble\Baballonia\src\Baballonia\Helpers\MainWindowExtensions.cs:line 11
   at Baballonia.App.<>c__DisplayClass6_2.<OnFrameworkInitializationCompleted>b__7(Object _, RoutedEventArgs _) in D:\Github-projects\Babble\Baballonia\src\Baballonia\App.axaml.cs:line 257
   at Avalonia.Interactivity.Interactive.<AddHandler>g__InvokeAdapter|8_0[TEventArgs](Delegate baseHandler, Object sender, RoutedEventArgs args)
   at Avalonia.Interactivity.Interactive.<>c__8`1.<AddHandler>b__8_1(Delegate baseHandler, Object sender, RoutedEventArgs args)
   at Avalonia.Interactivity.EventRoute.RaiseEventImpl(RoutedEventArgs e)
   at Avalonia.Interactivity.EventRoute.RaiseEvent(Interactive source, RoutedEventArgs e)
   at Avalonia.Interactivity.Interactive.RaiseEvent(RoutedEventArgs e)
   at Avalonia.Controls.Control.OnLoaded(RoutedEventArgs e)
   at Avalonia.Controls.Control.OnLoadedCore()
   at Avalonia.Controls.Control.<>c.<.cctor>b__70_0()
   at Avalonia.Threading.DispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.Win32.Win32DispatcherImpl.DispatchWorkItem()
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32DispatcherImpl.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.StartCore(String[] args)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, Action`1 lifetimeBuilder)
   at Baballonia.Desktop.Program.Main(String[] args) in D:\Github-projects\Babble\Baballonia\src\Baballonia.Desktop\Program.cs:line 30

Process finished with exit code -532,462,766.
```

# Default V1.0.0.1 Release installer

 - Launch installer
 - All Default options
 - Start program

### Expected behavior

Not crash on startup

### Operating System

Windows 11

### Hardware setup

Babble Tracker (wired)

### Log Files

[Log from stack trace example](https://github.com/user-attachments/files/22192672/From.stack.trace.example.baballonia_desktop.2025-09-06_21-43-53.log)

Note: all logs appear to end at that exact same spot regardless of installation or running method.

dfgHiatus, 2025-09-07:
Hello, have at any point you tried building from source with an older version of our source? You'll need to delete your local settings file in the apdata roaming folder

dfgHiatus, 2025-09-07:
Additionally, does this produce a log file under your Documents folder?

dfgHiatus, 2025-09-07:
This will produce a log file regardless of configuration.

dfgHiatus, 2025-09-07:
Also, please update to v1.1.0.2 and see if the issue persists.

virtuallyaverage, 2025-09-08:
> Hello, have at any point you tried building from source with an older version of our source? You'll need to delete your local settings file in the apdata roaming folder

I did that a while back and got a version around late July to build and run but it didn't move data to the VRCFT plugin.

dfgHiatus, 2025-09-08:
I think I might have a fix for this. Navigate to your AppData roaming folder, and delete the local settings json. Restart the app and let me know if the issue persists

dfgHiatus, 2025-09-08:
> My dumbass forgot completely that the roaming folder was a thing and it works now. Out of curiosity, is there a setting in particular that would likely is the culprit?
> 
> I'm looking at the file and nothing seems malformed.
> 
> Thanks for the help!

I think there's a sneaky race condition here somewhere. Might have to do with the camera feeds saving properly, but not being _loaded_ properly. Thanks for the report, and happy babbling!